Xinyuan-LilyGO/ESP32_S2 #1: [QUESTION] SPI PINS FOR TFT display

fermintm, 2021-06-16:
Hello. I am having trouble connecting an SPI display to this board. With other ESP32 or ESP8266 I do it without problem.
What are the exact pins? In the documentation I cannot clarify myself with the RESET, CS and DC pins.

Thanks

LilyGO, 2022-03-17:
Please send the detailed model number here, I will help you find the corresponding pin

educ8s, 2022-05-27:
I use this and it works.

mosi_pin = board.IO35
clk_pin = board.IO36
reset_pin = board.IO38
cs_pin = board.IO34
dc_pin = board.IO37

Shohreh, 2022-09-20:
I tried the above in Bodmer's User_Setup.h, [to no avail](https://github.com/Bodmer/TFT_eSPI/issues/2022).

Are there other files that need to be edited for TFT_eSPI to work on [this board](http://www.lilygo.cn/prod_view.aspx?TypeId=50062&Id=1321)?

lyusupov, 2022-10-08:
https://github.com/Bodmer/TFT_eSPI/issues/2022#issuecomment-1264568239