Xinyuan-LilyGO/ESP32_S2 #4: [T8 ESP32-S2 V1.1] TFT_eSPI won't work

Shohreh, 2022-09-20:
Hello,

I bought a Lilygo TTGO T8 ESP32-S2 V1.1 ST77789, but the display remains [desperately dark](https://github.com/Bodmer/TFT_eSPI/issues/2022).

What settings should I use for (Bodmer's) TFT_eSPI?

Thank you.

lyusupov, 2022-10-02:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Display-S2/blob/master/Arduino_Exapmles/libdeps/TFT_eSPI/User_Setups/Setup71_ESP32_S2_ST7789.h

Weird407, 2023-04-30:
I had a slightly different issue where I am able to light up the screen but I am unable to display any text,
I have selected the user template and board and changed the user_setup.h file to the same one as @lyusupov pointed to.
Everything works except for me being able to display text/images.
Any advice?