Xinyuan-LilyGO/ESP32_S2 #5: ESP32-S2-WROOM Pinout from the front

FabseRobotics, 2023-01-29:
Hey,
i use the ESP32-S2-WROOM in a breadboard, so i look all the time to the face of the board,
the board label is on the back, the pinout datasheet is from behind too,

so i draw a pinout from the front:
[ESP32-S2-WROOM.pdf](https://github.com/Xinyuan-LilyGO/ESP32_S2/files/10530986/ESP32-S2-WROOM.pdf)
[ESP32-S2-WROOM.pptx](https://github.com/Xinyuan-LilyGO/ESP32_S2/files/10530987/ESP32-S2-WROOM.pptx)

mrmonteith, 2023-03-25:
Thanks.  This was extremely helpful.  The upside down images drove me nuts. Of course they are all over the place.   Thanks again

Gabrieliess, 2025-01-23:
Finally normal usable pinout. May your bed pillow be cold for the rest of your life.